adafruit/RadioHead #4: Test RFM95 with nRF52, see what is amiss & fix!

ladyada, 2018-03-02:
https://forums.adafruit.com/viewtopic.php?f=57&t=117786&p=650818&e=650818

gojimmypi, 2018-03-03:
heads up that I've merged the latest 1.83 RadioHead into my AirSpayce branch and made some additional changes of my own - in particular to the RFM95 LoRa stuff (my target is ESP32 in the M5Stack). It seems to be working fairly well.

https://github.com/gojimmypi/RadioHead/tree/AirSpayce

Also added this suggestion on contributing:

https://github.com/gojimmypi/RadioHead/blob/AirSpayce/CONTRIBUTING.md

I've submitted a patch file from this branch to the author (Mike McCauley) and currently awaiting his review.

It would be great if all the great work from Adafruit could be kept in sync with the master zip:

http://www.airspayce.com/mikem/arduino/RadioHead/

For example one thing of interest recently added there is some encryption stuff.

ladyada, 2018-03-03:
oh yeah, thats a good point - we have to figure out a way to know when a new version is released from mikem. @gojimmypi wanna do a PR?

@tdicola when you look at this, update to the latest version of RadioHead and try out the Feather RXTX examples on a couple feather+radio kits

gojimmypi, 2018-03-03:
indeed. shy of hanging out in their forum, I don't think there's an easy way of finding out about their new releases.

https://groups.google.com/forum/#!forum/radiohead-arduino

Here's my PR: https://github.com/adafruit/RadioHead/pull/5 

In the meantime, fwiw - I had sporadic problems with my M5Stack display sharing the NSS line with the LoRa RA-02 module. I ended up needing to put LoRa to sleep whenever I used the display. (I didn't look to see if there are any similar slave select issues w/NRF52 feather + RFM95 lora)

I'm keeping track of my LoRa progress here:

https://gojimmypi.blogspot.com/2018/02/m5stack-lora-range-issues.html

Cheers

ladyada, 2018-04-12:
this is done! see latest commits :)

closed 2018-04-12 by ladyada