adafruit/RadioHead #6: How do I use a different hardware SPI interface if my board supports more than one?

TehProgrammer, 2018-03-27:
I am interested in using this library for a LoRa antenna tracker and that would require running two radios at once on two different SPI ports.  I have looked into it, and it seems there is a way to do it.  I just can't figure it out after looking at all the code...

gojimmypi, 2018-03-28:
With the RadioHead drivers, it is simply a matter of selecting the two key parameters (GPIO connection pins) during init: the CS (chip select) and INT (interrupt) pins specific to each device.

```
// Singleton instance of the radio driver
RH_RF95 rf95(RFM95_CS, RFM95_INT);
```

TehProgrammer, 2018-03-28:
Will this allow my to simultaneously read two radios at the same time?  I'm building an antenna tracker and it's important I can read the RSSI at about the same time for each radio.

gojimmypi, 2018-03-28:
well, "nearly simultaneously":  As they would both be interrupt driven, it would _appear_ they are reading concurrently. In reality, on a single core processor, only one interrupt would be processed at a time, Given the relatively low rates of data transfer vs processing speed - you'd likely never notice. If you are using something like FreeRTOS, it would appear even more like simultaneous events. (I've not actually used the RadioHead drivers with an RTOS yet)

ludoplt, 2019-02-28:
Hello, 

I want to use this library for a LORA project but we are using SAM D21E instead of SAM D21G.
SAN D21E only as 32 pins and don't have the pins for Arduino default hardware SPI. 

Is it possible to use this library but with another SPI interface ? 

Best regards, Ludovic

ludoplt, 2019-03-01:
We can change the ports using RHSoftwareSPI like this : 
```
#include <RHSoftwareSPI.h>
RHSoftwareSPI spi;
RH_RF69 rf69(RFM69_CS, RFM69_INT, spi);
void setup()
{
spi.setPins(12, 11, 13); // MISO 12, MOSI 11, SCK  13
}
```

RadioHead says this software SPI interface will be much slower than hardware SPI on most platforms... Perhaps not in SAM D21 ? 
Is there another way to achieve changing SPI port ?

orhanyor, 2019-05-01:
@ludoplt have you managed to find a way to use another SPI port for this library? im thinking may be a sercom SPI can be used if theres a constructor for it.

john--, 2019-08-10:
I'm also curious about setting up an additional SPI SERCOM for my SAMD21. Would be great if this was supported. I tried hacking something together with no luck.

Software SPI does seem to work fine, though.

john--, 2019-08-10:
Also, btw, it seems some boards support SPI1 and even SPI2 like the teensy.
For these you'd initial it like

```
#include <RHHardwareSPI1.h>
RH_NRF24 nrf24(9, 8, hardware_spi1);
```

Have it more generic to support custom SERCOMs would be great though.