adafruit/RadioHead #17: Encryption key NOT safe?!

StefanMeGit, 2018-11-19:
I made some test with two very short programs on the feather... the dont have the same ENCRYPTION KEY but the receiver is still getting  frequently messages from `rf69_manager.available()` ! Whats wrong here! Huge problem !! 

```
Message avaible!!
Sending failed (no ack)
Message avaible!!
Sending failed (no ack)
Message avaible!!
Sending failed (no ack)
Message avaible!!
Sending failed (no ack)
Message avaible!!
Sending failed (no ack)
Message avaible!!
Sending failed (no ack)
Message avaible!!
Sending failed (no ack)
Message avaible!!
Sending failed (no ack)
Message avaible!!
Sending failed (no ack)
Message avaible!!
Sending failed (no ack)
Message avaible!!
Sending failed (no ack)
```

TX:

```
// rf69 demo tx rx.pde
// -*- mode: C++ -*-
// Example sketch showing how to create a simple addressed, reliable messaging client
// with the RH_RF69 class. RH_RF69 class does not provide for addressing or
// reliability, so you should only use RH_RF69  if you do not need the higher
// level messaging abilities.
// It is designed to work with the other example rf69_server.
// Demonstrates the use of AES encryption, setting the frequency and modem
// configuration

#include <SPI.h>
#include <RH_RF69.h>
#include <RHReliableDatagram.h>
/************ Radio Setup ***************/

// Change to 434.0 or other frequency, must match RX's freq!
#define RF69_FREQ 433.0

// Where to send packets to!
#define DEST_ADDRESS   1
// change addresses for each client board, any number :)
#define MY_ADDRESS     2

  #define RFM69_CS      8
  #define RFM69_INT     3
  #define RFM69_RST     4
  #define LED           13

// Singleton instance of the radio driver
RH_RF69 rf69(RFM69_CS, RFM69_INT);

// Class to manage message delivery and receipt, using the driver declared above
RHReliableDatagram rf69_manager(rf69, MY_ADDRESS);

struct package {
  uint8_t value1 = 1;
  uint16_t value2 = 222;
  uint8_t value3 = 3;
  uint8_t value4 = 4;
} remPackage;

struct callback {
  float value1 = 100.0;
  float value2 = 200.0;
  long value3 = 300;
  long value4 = 400;
  uint8_t value5 = 5;
  float value6 = 600.0;
  float value7 = 700.0;
  float value8 = 800.0;
  bool value9 = true;
} returnData;

void setup()
{
  Serial.begin(115200);
  //while (!Serial) { delay(1); } // wait until serial console is open, remove if not tethered to computer

  pinMode(LED, OUTPUT);
  pinMode(RFM69_RST, OUTPUT);
  digitalWrite(RFM69_RST, LOW);

  Serial.println("Feather Addressed RFM69 TX Test!");
  Serial.println();

  // manual reset
  digitalWrite(RFM69_RST, HIGH);
  delay(10);
  digitalWrite(RFM69_RST, LOW);
  delay(10);

  if (!rf69_manager.init()) {
    Serial.println("RFM69 radio init failed");
    while (1);
  }
  Serial.println("RFM69 radio init OK!");
  // Defaults after init are 434.0MHz, modulation GFSK_Rb250Fd250, +13dbM (for low power module)
  // No encryption
  if (!rf69.setFrequency(RF69_FREQ)) {
    Serial.println("setFrequency failed");
  }

  // If you are using a high power RF69 eg RFM69HW, you *must* set a Tx power with the
  // ishighpowermodule flag set like this:
  rf69.setTxPower(20, true);  // range from 14-20 for power, 2nd arg must be true for 69HCW

  // The encryption key has to be the same as the one in the server
  uint8_t key[] = { 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08,
                    0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x02};
  rf69.setEncryptionKey(key);

  pinMode(LED, OUTPUT);

  Serial.print("RFM69 radio @");  Serial.print((int)RF69_FREQ);  Serial.println(" MHz");
}

// Dont put this on the stack:

uint8_t len = sizeof(remPackage);

void loop() {
  delay(50);  // Wait 50 ms between transmits, could also 'sleep' here!

  // Send a message to the DESTINATION!
  if (rf69_manager.sendtoWait((uint8_t *)&remPackage, len, DEST_ADDRESS)) {
    // Now wait for a reply from the server
    len = sizeof(returnData);
    uint8_t from;
    if (rf69_manager.recvfromAckTimeout((uint8_t *)&returnData, &len, 50, &from)) {

      Serial.print("Got reply from #"); Serial.print(from);
      Serial.print(" [RSSI :");
      Serial.print(rf69.lastRssi());
      Serial.print("] : ");
      Serial.print("value1 ");Serial.println(returnData.value1);
      Serial.print("value2 ");Serial.println(returnData.value2);
      Serial.print("value3 ");Serial.println(returnData.value3);
      Serial.print("value4 ");Serial.println(returnData.value4);
    } else {
      Serial.println("No reply, is anyone listening?");
    }
  } else {
    Serial.println("Sending failed (no ack)");
  }
}
```

RX:

```
// rf69 demo tx rx.pde
// -*- mode: C++ -*-
// Example sketch showing how to create a simple addressed, reliable messaging client
// with the RH_RF69 class. RH_RF69 class does not provide for addressing or
// reliability, so you should only use RH_RF69  if you do not need the higher
// level messaging abilities.
// It is designed to work with the other example rf69_server.
// Demonstrates the use of AES encryption, setting the frequency and modem
// configuration

#include <SPI.h>
#include <RH_RF69.h>
#include <RHReliableDatagram.h>

/************ Radio Setup ***************/

// Change to 434.0 or other frequency, must match RX's freq!
#define RF69_FREQ 433.0

// who am i? (server address)
#define MY_ADDRESS     1

  #define RFM69_CS      8
  #define RFM69_INT     3
  #define RFM69_RST     4

// Singleton instance of the radio driver
RH_RF69 rf69(RFM69_CS, RFM69_INT);

// Class to manage message delivery and receipt, using the driver declared above
RHReliableDatagram rf69_manager(rf69, MY_ADDRESS);

struct package {
  uint8_t value1 = 1;
  uint16_t value2 = 222;
  uint8_t value3 = 3;
  uint8_t value4 = 4;
} remPackage;

struct callback {
  float value1 = 100.0;
  float value2 = 200.0;
  long value3 = 300;
  long value4 = 400;
  uint8_t value5 = 5;
  float value6 = 600.0;
  float value7 = 700.0;
  float value8 = 800.0;
  bool value9 = true;
} returnData;

void setup()
{
  Serial.begin(115200);
  while (!Serial) { delay(1); } // wait until serial console is open, remove if not tethered to computer

  pinMode(RFM69_RST, OUTPUT);
  digitalWrite(RFM69_RST, LOW);

  Serial.println("Feather Addressed RFM69 RX Test!");
  Serial.println();

  // manual reset
  digitalWrite(RFM69_RST, HIGH);
  delay(10);
  digitalWrite(RFM69_RST, LOW);
  delay(10);

  if (!rf69_manager.init()) {
    Serial.println("RFM69 radio init failed");
    while (1);
  }
  Serial.println("RFM69 radio init OK!");
  // Defaults after init are 434.0MHz, modulation GFSK_Rb250Fd250, +13dbM (for low power module)
  // No encryption
  if (!rf69.setFrequency(RF69_FREQ)) {
    Serial.println("setFrequency failed");
  }

  // If you are using a high power RF69 eg RFM69HW, you *must* set a Tx power with the
  // ishighpowermodule flag set like this:
  rf69.setTxPower(20, true);  // range from 14-20 for power, 2nd arg must be true for 69HCW

  // The encryption key has to be the same as the one in the server
  uint8_t key[] = { 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08,
                    0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x01};
  rf69.setEncryptionKey(key);

  Serial.print("RFM69 radio @");  Serial.print((int)RF69_FREQ);  Serial.println(" MHz");
}

// Dont put this on the stack:
uint8_t data[] = "And hello back to you";
// Dont put this on the stack:

void loop() {
  if (rf69_manager.available())
  {
    Serial.println("Message avaible!!");
    // Wait for a message addressed to us from the client
    uint8_t len = sizeof(remPackage);
    uint8_t from;
    if (rf69_manager.recvfromAckTimeout((uint8_t*)&remPackage, (uint8_t*)&len, 50, &from)) {
      if (returnData.value1 > 1 || returnData.value2 > 222 || returnData.value3 > 3 || returnData.value4 > 4){
        Serial.print("Got reply from #"); Serial.print(from);
        Serial.print(" [RSSI :");
        Serial.print(rf69.lastRssi());
        Serial.println("] : ");
        Serial.print("value1: ");Serial.println(returnData.value1);
        Serial.print("value2: ");Serial.println(returnData.value2);
        Serial.print("value3: ");Serial.println(returnData.value3);
        Serial.print("value4: ");Serial.println(returnData.value4);
        }
      }

      // Send a reply back to the originator client
      len = sizeof(returnData);
      if (!rf69_manager.sendtoWait((uint8_t*)&returnData, len, from)){
        Serial.println("Sending failed (no ack)");
    }
  }
}

```

IoTPanic, 2020-10-04:
@StefanMeGit in your following code you receive a message, but you are not printing what you received vs what you are transmitting, if you send an encrypted message, anyone can still receive it with `rf69_manager.available()` but Im willing to bet the message you get is garbled and not the same thing you transmitted.

IoTPanic, 2020-10-04:
This is old, if what I am saying is true, or you don't care anymore, this can be closed