adafruit/RadioHead #28: No M4 support (ATSAMD51)

IoTPanic, 2019-01-24:
Hello,

I was just compiling the RadioHead library for the Adafruit feather M4 and I am unable to compile as the library only supports the ATSAMD21's timers and not the ATSAMD51s. The problem is in RH_ASK.cpp on the fallowing lines-
```
#elif defined (__arm__) && defined(ARDUINO_ARCH_SAMD)
    // Arduino Zero
    #define RH_ASK_ZERO_TIMER TC3
    // Clock speed is 48MHz, prescaler of 64 gives a good range of available speeds vs precision
    #define RH_ASK_ZERO_PRESCALER 64
    #define RH_ASK_ZERO_TIMER_IRQ TC3_IRQn

    // Enable clock for TC
    REG_GCLK_CLKCTRL = (uint16_t) (GCLK_CLKCTRL_CLKEN | GCLK_CLKCTRL_GEN_GCLK0 | GCLK_CLKCTRL_ID(GCM_TCC2_TC3)) ;
    while ( GCLK->STATUS.bit.SYNCBUSY == 1 ); // wait for sync
    
    // The type cast must fit with the selected timer mode
    TcCount16* TC = (TcCount16*)RH_ASK_ZERO_TIMER; // get timer struct
    
    TC->CTRLA.reg &= ~TC_CTRLA_ENABLE;   // Disable TC
    while (TC->STATUS.bit.SYNCBUSY == 1); // wait for sync
    
    TC->CTRLA.reg |= TC_CTRLA_MODE_COUNT16;  // Set Timer counter Mode to 16 bits
    while (TC->STATUS.bit.SYNCBUSY == 1); // wait for sync
    TC->CTRLA.reg |= TC_CTRLA_WAVEGEN_MFRQ; // Set TC as Match Frequency
    while (TC->STATUS.bit.SYNCBUSY == 1); // wait for sync

    // Compute the count required to achieve the requested baud (with 8 interrupts per bit)
    uint32_t rc = (VARIANT_MCK / _speed) / RH_ASK_ZERO_PRESCALER / 8;
    
    TC->CTRLA.reg |= TC_CTRLA_PRESCALER_DIV64;   // Set prescaler to agree with RH_ASK_ZERO_PRESCALER
    while (TC->STATUS.bit.SYNCBUSY == 1); // wait for sync
    
    TC->CC[0].reg = rc; // FIXME
    while (TC->STATUS.bit.SYNCBUSY == 1); // wait for sync
    
    // Interrupts
    TC->INTENSET.reg = 0;              // disable all interrupts
    TC->INTENSET.bit.MC0 = 1;          // enable compare match to CC0
    
    // Enable InterruptVector
    NVIC_ClearPendingIRQ(RH_ASK_ZERO_TIMER_IRQ);
    NVIC_EnableIRQ(RH_ASK_ZERO_TIMER_IRQ);
    
    // Enable TC
    TC->CTRLA.reg |= TC_CTRLA_ENABLE;
    while (TC->STATUS.bit.SYNCBUSY == 1); // wait for sync
```

Does anyone have the correct file that could be used with the M4? In case you need the compilation error-
```
C:\Users\woode\.platformio\lib\RadioHead_ID124/RadioHead.h:306:51: warning: "/*" within comment [-Wcomment]
cp /usr/local/projects/arduino/libraries/RadioHead/*.cpp .
^
C:\Users\woode\.platformio\lib\RadioHead_ID124\RH_ASK.cpp: In member function 'void RH_ASK::timerSetup()':
C:\Users\woode\.platformio\lib\RadioHead_ID124\RH_ASK.cpp:246:5: error: 'REG_GCLK_CLKCTRL' was not declared in this scope
REG_GCLK_CLKCTRL = (uint16_t) (GCLK_CLKCTRL_CLKEN | GCLK_CLKCTRL_GEN_GCLK0 | GCLK_CLKCTRL_ID(GCM_TCC2_TC3)) ;
^
C:\Users\woode\.platformio\lib\RadioHead_ID124\RH_ASK.cpp:246:36: error: 'GCLK_CLKCTRL_CLKEN' was not declared in this scope
REG_GCLK_CLKCTRL = (uint16_t) (GCLK_CLKCTRL_CLKEN | GCLK_CLKCTRL_GEN_GCLK0 | GCLK_CLKCTRL_ID(GCM_TCC2_TC3)) ;
^
C:\Users\woode\.platformio\lib\RadioHead_ID124\RH_ASK.cpp:246:57: error: 'GCLK_CLKCTRL_GEN_GCLK0' was not declared in this scope
REG_GCLK_CLKCTRL = (uint16_t) (GCLK_CLKCTRL_CLKEN | GCLK_CLKCTRL_GEN_GCLK0 | GCLK_CLKCTRL_ID(GCM_TCC2_TC3)) ;
^
C:\Users\woode\.platformio\lib\RadioHead_ID124\RH_ASK.cpp:246:110: error: 'GCLK_CLKCTRL_ID' was not declared in this scope
REG_GCLK_CLKCTRL = (uint16_t) (GCLK_CLKCTRL_CLKEN | GCLK_CLKCTRL_GEN_GCLK0 | GCLK_CLKCTRL_ID(GCM_TCC2_TC3)) ;
^
C:\Users\woode\.platformio\lib\RadioHead_ID124\RH_ASK.cpp:247:19: error: 'struct Gclk' has no member named 'STATUS'
while ( GCLK->STATUS.bit.SYNCBUSY == 1 ); // wait for sync
^
C:\Users\woode\.platformio\lib\RadioHead_ID124\RH_ASK.cpp:253:27: error: 'volatile struct TC_STATUS_Type::<anonymous>' has no member named 'SYNCBUSY'
while (TC->STATUS.bit.SYNCBUSY == 1); // wait for sync
^
C:\Users\woode\.platformio\lib\RadioHead_ID124\RH_ASK.cpp:256:27: error: 'volatile struct TC_STATUS_Type::<anonymous>' has no member named 'SYNCBUSY'
while (TC->STATUS.bit.SYNCBUSY == 1); // wait for sync
^
C:\Users\woode\.platformio\lib\RadioHead_ID124\RH_ASK.cpp:257:22: error: 'TC_CTRLA_WAVEGEN_MFRQ' was not declared in this scope
TC->CTRLA.reg |= TC_CTRLA_WAVEGEN_MFRQ; // Set TC as Match Frequency
^
C:\Users\woode\.platformio\lib\RadioHead_ID124\RH_ASK.cpp:258:27: error: 'volatile struct TC_STATUS_Type::<anonymous>' has no member named 'SYNCBUSY'
while (TC->STATUS.bit.SYNCBUSY == 1); // wait for sync
^
C:\Users\woode\.platformio\lib\RadioHead_ID124\RH_ASK.cpp:264:27: error: 'volatile struct TC_STATUS_Type::<anonymous>' has no member named 'SYNCBUSY'
while (TC->STATUS.bit.SYNCBUSY == 1); // wait for sync
^
C:\Users\woode\.platformio\lib\RadioHead_ID124\RH_ASK.cpp:267:27: error: 'volatile struct TC_STATUS_Type::<anonymous>' has no member named 'SYNCBUSY'
while (TC->STATUS.bit.SYNCBUSY == 1); // wait for sync
^
C:\Users\woode\.platformio\lib\RadioHead_ID124\RH_ASK.cpp:279:27: error: 'volatile struct TC_STATUS_Type::<anonymous>' has no member named 'SYNCBUSY'
while (TC->STATUS.bit.SYNCBUSY == 1); // wait for sync
```

Thank you,
Sam

ladyada, 2019-01-24:
hmm, we dont use ASK so that file doesn't need to compile/work
wanna try updating the #ifdef line to  have something like `&& ! defined(__SAMD51__)` (not in front of comptuer so just check the compile line for the right __token__ and then see if it compiles for ya?

IoTPanic, 2019-01-25:
Hey LadyAda! Its a privilege to talk to you!

I made a pull request that solves the issue temporarily. Adding the line you suggested did not work, however, my fix allows for all ATSAMD51 boards to compile any other driver but continues to not support RH_ASK for the ATSAMD51. I think I have found the fix, I will work at it this weekend and I've figured out all the timer values as well as found how to set up a TC so I will try to add support. It seems the compiler was including RH_ASK.h for some reason, I am very confused on it, and went through all the header files I included and none say anything about that file. I originally thought it was a PlatformIO mistake.

Should I keep the issue open until I find a fix or no? I don't know if not being able to use a Cortex-M4 and a ASK transceiver would be considered an "issue".

WunderBeard, 2020-10-03:
> I think I have found a solution to support ASK, I will work at it this weekend and I've figured out all the timer values as well as found how to set up a TC so I will try to add support (For fun).
> ...
> Should I keep the issue open until I find a fix or no? I don't know if not being able to use a Cortex-M4 and a ASK transceiver would be considered an "issue".

@IoTPanic, I am having this very issue of trying to use a cheap ASK transceiver with an ItsyBitsy M4. Did you ever successfully create the correct timer settings for the SAMD51? It sure would save me a lot of trouble as I'm new to the M4.

It seems issue #15 is also about this same thing (and it remains open).

Any help would be greatly appreciated. Thanks!

IoTPanic, 2020-10-03:
I did, here is the PR- #29

IoTPanic, 2020-10-03:
Fix is certainly possible, let me get some sleep and I'll add a bit in for enabling the SAMD51 clock IRQ requires for RGH_ASK

IoTPanic, 2020-10-04:
@WunderBeard could you please try this out for me so I can make a pull request if it works? Have not really been able to test but it compiles and _should_ work - https://github.com/IoTPanic/RadioHead/tree/patch-m4_RH_ASK-iotpanic

WunderBeard, 2020-10-05:
@IoTPanic, thank you for your quick response! It may take me a day or three to get the physical setup running (in my precious spare time). I'm swapping out an Arduino Nano in favor of the ItsyBitsy M4 Express, so I have some wiring and pinout checking to do. The compile error was my first stumbling block. Maybe I can get a simple Hello World running just to check the radios. Your code from #58 does compile for me too, so I have high hopes. Sorry I can't offer a faster turn-around.

IoTPanic, 2020-10-05:
Well sorry for that compile error @WunderBeard, looking back [I actually created that](https://github.com/adafruit/RadioHead/issues/28#issuecomment-457413281), this RH_ASK file broke the entire library if using the SAMD51, and I made a fix last year to let everything else compile, saying I would come back and fix this and never did. I could have at least added some preprocessor conditions to at least give you a readable error. Hope I  didn't turn you off from the M4 itsy bitsy, I love that platform and highly recommend it.

If you are able to run a hello world, that would be very helpful to verify everything works for the next guy.

IoTPanic, 2020-10-09:
@ladyada issue fixed and fix is confirmed working in #58, can be closed, going to wait for you to review or PR to be accepted.

IoTPanic, 2021-03-01:
Anyone able to check, test, merge?

veggie2u, 2021-09-11:
Did this get merged, but the issue is open, or did it not get merged?

IoTPanic, 2021-09-11:
@veggie2u it is not merged, according to github since my last push on my fork there hasn't been any activity, I would suggest you use that until someone gives the repo a little love.

https://github.com/IoTPanic/RadioHead/tree/patch-m4_RH_ASK-iotpanic

veggie2u, 2021-09-13:
Dang! Thanks @IoTPanic It works just fine. @jwcooper @ladyada I have tested their branch on an ItsyBitsy M0, and a M4, and a Feather M0, and M4. All work using the default RadioHead69_AddrDemo_RX and RadioHead69_addrDemo_Tx (with only changes being pins)