adafruit/RadioHead #37: Need help with receiving boiler thermostat messages

mihai-dinculescu, 2020-01-03:
Hello, I would really appreciate any help with my little project.

Hardware:
- Danfoss-RX1 RFM69 433MHz boiler radio receiver https://manuall.co.uk/danfoss-rx1-thermostat/
- NodeMCU ESP 12-E
- Adafruit RFM69HCW and RFM9X LoRa Packet Radio Breakout

What am I trying to achieve:
- Receive messages that are sent from the Thermostat to the boiler Receiver.
- I found this example of someone else making it work with a custom firmware, but I'm not advanced enough to understand much of it: https://github.com/adpeace/thermostat/blob/master/rf69.cpp

Current situation:
- RFM69 seems to init OK but no messages are received

My code so far:
```
#include <stdint.h>
#include <sys/time.h>

#include <Arduino.h>

#include <SPI.h>
#include <RH_RF69.h>

#define RF69_FREQ 433.92

// NodeMCU ESP 12-E
#define LED           2
#define RFM69_CS      D8
#define RFM69_IRQ     D1
#define RFM69_RST     D2
#define RFM69_INT     D1

// Singleton instance of the radio driver
RH_RF69 rf69(RFM69_CS, RFM69_INT);

int16_t packetnum = 0;  // packet counter, we increment per xmission

void setup() {
  Serial.begin(115200);

  pinMode(LED, OUTPUT);     
  pinMode(RFM69_RST, OUTPUT);
  digitalWrite(RFM69_RST, LOW);

  Serial.println();
  Serial.println("RFM69 RX Test!");
  Serial.println();

  // manual reset
  digitalWrite(RFM69_RST, HIGH);
  delay(10);
  digitalWrite(RFM69_RST, LOW);
  delay(10);
  
  if (!rf69.init()) {
    Serial.println("RFM69 radio init failed");
    while (1);
  }
  Serial.println("RFM69 radio init OK!");
  
  // Defaults after init are 434.0MHz, modulation GFSK_Rb250Fd250, +13dbM (for low power module)
  // No encryption
  if (!rf69.setFrequency(RF69_FREQ)) {
    Serial.println("setFrequency failed");
  }

  // If you are using a high power RF69 eg RFM69HW, you *must* set a Tx power with the
  // ishighpowermodule flag set like this:
  rf69.setTxPower(14, true);  // range from 14-20 for power, 2nd arg must be true for 69HCW
}

void loop() {
 if (rf69.available()) {
    Serial.printf("There is a message!");
    // Should be a message for us now   
    uint8_t buf[RH_RF69_MAX_MESSAGE_LEN];
    uint8_t len = sizeof(buf);
    if (rf69.recv(buf, &len)) {
      if (!len) return;
      buf[len] = 0;
      Serial.print("Received [");
      Serial.print(len);
      Serial.print("]: ");
      Serial.println((char*)buf);
      Serial.print("RSSI: ");
      Serial.println(rf69.lastRssi(), DEC);
    } else {
      Serial.println("Receive failed");
    }
  }
}

int64_t GetTimestamp() {
  struct timeval tv;
  gettimeofday(&tv, NULL);
  return (tv.tv_sec * 1000LL + (tv.tv_usec / 1000LL));
}
```

mihai-dinculescu, 2020-01-05:
It turns out that I had to ditch the RF library and read the raw bytes out of FIFO.

closed 2020-01-05 by mihai-dinculescu

mkarklins, 2020-12-12:
@mihai-dinculescu would you mind sharing a sample code for that?

mihai-dinculescu, 2020-12-12:
> 
> 
> @mihai-dinculescu would you mind sharing a sample code for that?

Here you go, here's my working project. I hope that it's gonna be useful to you.
https://github.com/mihai-dinculescu/home-automation-thermostat