adafruit/RadioHead #39: RH_ASK: Guru Meditation Error: Core 1 panic'ed (Cache disabled but cached memory region accessed) ESP32 VSC Platformio

petervflocke, 2020-01-12:
RH_ASK together with WiFi in some cases leads to Guru Meditation Error (Cache disabled but cached memory region accessed)
Solution: add ESP32 handling 
`#define INTERRUPT_ATTR ICACHE_RAM_ATTR`
by changing in RH_ASK.cpp file 

### FROM:
```cpp
#if (RH_PLATFORM == RH_PLATFORM_ESP8266)
    // interrupt handler and related code must be in RAM on ESP8266,
    // according to issue #46.
    #define INTERRUPT_ATTR ICACHE_RAM_ATTR
#else
    #define INTERRUPT_ATTR
#endif
```
### TO:
```cpp
#if (RH_PLATFORM == RH_PLATFORM_ESP8266)
    // interrupt handler and related code must be in RAM on ESP8266,
    // according to issue #46.
    #define INTERRUPT_ATTR ICACHE_RAM_ATTR
#elif (RH_PLATFORM == RH_PLATFORM_ESP32)
    #define INTERRUPT_ATTR IRAM_ATTR 
#else
    #define INTERRUPT_ATTR
#endif
```

If this is a correct way to solve it, please include in your next version.
By the way I do expect same problem (and solution) for the module CC110, which I want o use in the next (hardware) release in my project. 

Best
Peter

petervflocke, 2020-01-19:
Update:
I needed to add `DRAM_ATTR` to the table `symbols[]` definition. This and the latter change are done in RH_ASK.cpp.

```cpp
// :PP: added DRAM_ATTR
#if (RH_PLATFORM == RH_PLATFORM_ESP8266)
    #define D_A DRAM_ATTR
#elif (RH_PLATFORM == RH_PLATFORM_ESP32)
    #define D_A DRAM_ATTR  
#else
    #define D_A 
#endif

D_A static uint8_t symbols[] =
{
    0xd,  0xe,  0x13, 0x15, 0x16, 0x19, 0x1a, 0x1c, 
    0x23, 0x25, 0x26, 0x29, 0x2a, 0x2c, 0x32, 0x34
};
```
Kindly ask you to verify and confirm that this is fine from your point of view. After both changes I haven't observed any further guru mediation.

rkopka42, 2020-02-18:
The DRAM_ATTR part was what really helped me. The IRAM part was already inside the lib, but I still got the Cache panics regularily and I only looked inside the code part.