adafruit/RadioHead #45: Register Number Bug in RFM95 for RegNbRxBytes

wardjm, 2020-04-10:
I'm looking at: https://cdn.sparkfun.com/assets/learn_tutorials/8/0/4/RFM95_96_97_98W.pdf (page 85).

The code in RH_RF95.h shows:
#define RH_RF95_REG_13_RX_NB_BYTES                         0x13
but the document says:
0x1D RegNbRxBytes  Number of received bytes

So, the code seems to be reading IF Frequency LSB (reg 0x13) instead of number of bytes received. I haven't noticed any bugs due to this, but I also don't know for sure whether the documentation is wrong or the code is wrong. You'd think if the code were wrong, there would be more issues since it is a crucial piece of information.

gojimmypi, 2020-04-10:
Interesting. Check out the [Semtech docs](https://www.semtech.com/products/wireless-rf/lora-transceivers/sx1276#download-resources) for the SX1276/77/78/79 (I believe the SX1276 is the RFM95 equivalent) on pages 90 and 91: 

[image]

It was my understanding they were equivalents.

wardjm, 2020-04-10:
Well that could explain why it seems to work! Maybe the RFM95 docs are wrong somehow?