adafruit/RadioHead #65: RFM95 with nRF52840 issue

Ryann98, 2022-02-28:
I have an nRF52840 feather and a Lora Wing RFM95W stacked on top of it.  I am unable to  run any of the example codes in the RadioHead library such as "Feather9x_TX"  as they're failing to compile 
Looking at the code of the RadioHead library, there is ARDUINO_ARCH_NRF52 which makes you think it supports the NRF52 chips but it still tries to do the things and fails :

`C:\Users\Rayan\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp: In member function 'virtual void RHHardwareSPI::begin()':
C:\Users\Rayan\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:104:39: error: invalid conversion from 'uint8_t' {aka 'unsigned char'} to 'BitOrder' [-fpermissive]
  104 |    _settings = SPISettings(frequency, bitOrder, dataMode);
      |                                       ^~~~~~~~
      |                                       |
      |                                       uint8_t {aka unsigned char}
In file included from C:\Users\Rayan\Documents\Arduino\libraries\RadioHead/RadioHead.h:1181,
                 from C:\Users\Rayan\Documents\Arduino\libraries\RadioHead/RHGenericSPI.h:10,
                 from C:\Users\Rayan\Documents\Arduino\libraries\RadioHead/RHHardwareSPI.h:10,
                 from C:\Users\Rayan\Documents\Arduino\libraries\RadioHead\RHHardwareSPI.cpp:7:
C:\Users\Rayan\Documents\ArduinoData\packages\adafruit\hardware\nrf52\1.3.0\libraries\SPI/SPI.h:42:42: note:   initializing argument 2 of 'SPISettings::SPISettings(uint32_t, BitOrder, uint8_t)'
   42 |     SPISettings(uint32_t clock, BitOrder bitOrder, uint8_t dataMode) {
      |                                 ~~~~~~~~~^~~~~~~~
exit status 1
Error compiling for board Adafruit Feather nRF52840 Sense.
`

Ryann98, 2022-03-01:
Sorry I think I forgot to mention something. The above error was obtained when I removed the code for board definitions and defined my own set of pins.
If I keep the code in "Feather9x_TX"  as it is, I get the following error:

`Feather9x_TX:77:14: error: 'RFM95_CS' was not declared in this scope
   77 | RH_RF95 rf95(RFM95_CS, RFM95_INT);
      |              ^~~~~~~~
Feather9x_TX:77:24: error: 'RFM95_INT' was not declared in this scope
   77 | RH_RF95 rf95(RFM95_CS, RFM95_INT);
      |                        ^~~~~~~~~
C:\Users\Rayan\Documents\Arduino\libraries\RadioHead\examples\feather\Feather9x_TX\Feather9x_TX.ino: In function 'void setup()':
Feather9x_TX:81:11: error: 'RFM95_RST' was not declared in this scope
   81 |   pinMode(RFM95_RST, OUTPUT);
      |           ^~~~~~~~~
exit status 1
'RFM95_CS' was not declared in this scope
`
Seems like define(nRF52) is not working and hence why I had to define my own pins as I have them connected on my board:
`  #define RFM95_RST   11   // "A"
  #define RFM95_CS      10   // "B"
  #define RFM95_INT     9   // "C"
  `

Ryann98, 2022-03-01:
Hello Ha Thach, 
Thank you for your prompt response. Sure, please find attached the whole compile log and the code that I am using for testing.
[Compiling Error.txt](https://github.com/adafruit/RadioHead/files/8163627/Compiling.Error.txt)
[Feather9x_TX_test2.txt](https://github.com/adafruit/RadioHead/files/8163628/Feather9x_TX_test2.txt)

hathach, 2022-03-03:
an fix is on the way https://github.com/adafruit/RadioHead/pull/66 , please wait until it is reviewed/merged then we could roll out a new release for the fix.

hathach, 2022-03-10:
fixed by #66

closed 2022-03-10 by hathach