adafruit/RadioHead #79: Won't Compile (using ESP32 Feather with RFM69HCW wing)

chuckberrypi, 2023-12-31:
After striking out getting the LowPowerLabs RFM69 library working on this hardware setup, I thought I'd switch tacks and try RadioHead. I can't even get it to compile. Here's the error messages I get when I try to compile: 

```~/Documents/Arduino/libraries/RadioHead-master/RH_ASK.cpp: In member function 'void RH_ASK::timerSetup()':
~/Documents/Arduino/libraries/RadioHead-master/RH_ASK.cpp:528:23: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
  528 |     timer = timerBegin(0, 80, true); // Alarm value will be in in us
      |             ~~~~~~~~~~^~~~~~~~~~~~~
In file included from ~/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32/esp32-hal.h:84,
                 from ~/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32/Arduino.h:36,
                 from ~/Documents/Arduino/libraries/RadioHead-master/RadioHead.h:1544,
                 from ~/Documents/Arduino/libraries/RadioHead-master/RHGenericDriver.h:9,
                 from ~/Documents/Arduino/libraries/RadioHead-master/RH_ASK.h:9,
                 from ~/Documents/Arduino/libraries/RadioHead-master/RH_ASK.cpp:6:
~/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32/esp32-hal-timer.h:35:14: note: declared here
   35 | hw_timer_t * timerBegin(uint32_t frequency);
      |              ^~~~~~~~~~
~/Documents/Arduino/libraries/RadioHead-master/RH_ASK.cpp:529:25: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
  529 |     timerAttachInterrupt(timer, &esp32_timer_interrupt_handler, true);
      |     ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t * timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
~/Documents/Arduino/libraries/RadioHead-master/RH_ASK.cpp:530:5: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  530 |     timerAlarmWrite(timer, 1000000 / _speed / 8, true);
      |     ^~~~~~~~~~~~~~~
      |     timerWrite
~/Documents/Arduino/libraries/RadioHead-master/RH_ASK.cpp:531:5: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  531 |     timerAlarmEnable(timer);
      |     ^~~~~~~~~~~~~~~~
      |     timerAlarm

exit status 1

Compilation error: exit status 1```

Any suggestions for getting this off the ground? Do I need to change the library code? Pretty frustrating that I can't get the wing to work with the ESP32 feather.

AniSerNik, 2024-02-04:
Hi. This happened because ESPRESSIF changed the Timer API in version 3.0.0. More about the situation here:
https://github.com/espressif/arduino-esp32/issues/8796
As a temporary solution, I changed lines 563-566 in RH_ASK.cpp . I cannot claim that this is absolutely true, but I have not found any errors in my projects. Tested on RH_RF95.
Old code:
`timer = timerBegin(0, 80, true); // Alarm value will be in in us
 timerAttachInterrupt(timer, &esp32_timer_interrupt_handler, true);
 timerAlarmWrite(timer, 1000000 / _speed / 8, true);
 timerAlarmEnable(timer);`
New code:
`timer = timerBegin(1000000);
timerAttachInterrupt(timer, &esp32_timer_interrupt_handler);
timerAlarm(timer, 1000000 / _speed / 8, true, 0);`

StillVoidingWarranties, 2024-04-23:
Just encountered the same problem and the changed timer call from @AniSerNik worked. Thank you very much.

Osmoztech-io, 2024-10-27:
hw_timer_t *timer = timerBegin(1000000);  
timerAttachInterrupt(timer, &onTimer); 
timerAlarm(timer, 10000, true, 0);